berarma/new-lg4ff #116: Can load hid-logitech-new unless I blacklist hid-logitech

EazyDefcon, 2025-01-05:
Hi!
I have to blacklist hid-logitech (that comes with my kernel) to get new-lg4ff (hid-logitech-new) to load at boot. 

If I want to do it manually I can also do:
```
sudo rmmod hid-logitech

sudo modprobe hid-logitech-new
```

I'm unsure if its the correct way to make this to work?
I'm on Debian 12 stable

berarma, 2025-02-05:
Are you using `dkms`? If you do it will replace the module in the kernel.

You might need to update initramfs to replace the modules loaded on boot. Instructions are in the README.

EazyDefcon, 2025-02-07:
Yes I installed using dkms and updated initramfs as the readme says. I have to blacklist the old module or the hid-logitech-new wont load at boot. 

Perhaps I have done something wrong, but It does not matter for me as it all work as it should. 

I was hoping to have the leds on the wheel to work as intended for Assetto Corsa Competizione and I guess I need new-lg4ff and [this](https://github.com/d4rk/ac_shifting_leds) to make that work. Have not yet come around installing this...

Thank you for your work!!

berarma, 2025-02-07:
Ok. There are no more cases that I know, so I'll close it.

closed 2025-02-07 by berarma