berarma/new-lg4ff #118: Make pedals settings much more flexible

lestcape, 2025-01-11:
Is there any way to define the pedal we want to use or to detect the pedals on the fly? The issue would be to allow the pedals to be from another brand or they can be from the same brand but be connected as an independent device. The latter can be achieved for example in the Logitech G29 by buying this cable: https://www.aliexpress.com/item/1005005874201800.html?spm=a2g0o.order_list.order_list_main.467.3f15194dEmpbb6 When is connected it see that way:
[image]

What I would like to happen is that regardless of whether the pedal is recognized as a different device, the driver should map its functions as if they were part of the same device, so that everything is mapped to a single device. This would help it to be used as a whole in programs where using more than one device at a time is not allowed.

It may be possible to find an isolated pedal by finding devices that have only 3 axes.

lestcape, 2025-01-11:
If you want to know the reason for the Aliexpress cable, it is because it is especially useful so that the pedal cable does not get caught in the middle of your configuration and thus prevent it from being damaged.

berarma, 2025-01-11:
This is too much for a kernel driver. I think there might already exist some application doing this using virtual devices. Even if it's an application for joysticks it should work for wheels.

closed 2025-01-11 by berarma

lestcape, 2025-01-11:
I highly doubt that the app you're thinking of, aside from merging the controls, could somehow figure out how to create a new device with Force Feedback support. That's the real reason I thought that here, on the controller, would be where something like this could be done best and easiest.

Anyway, it's clear that for you this is out of scope of your driver. Have a good day and really thanks for the driver.

writequit, 2025-01-11:
Have a look at protopedal, I use it for other reasons but it does seem to know about ffb.

berarma, 2025-01-11:
> I highly doubt that the app you're thinking of, aside from merging the controls, could somehow figure out how to create a new device with Force Feedback support. That's the real reason I thought that here, on the driver, would be where something like this could be done best and easiest.
> 
> Anyway, it's clear that for you this is out of scope of your driver. Have a good day and really thanks for the driver.

It wouldn't be easy at all, at least for me. I don't even know if that's possible. All modules I know handle just one device. Sorry.

For the computer, a wheel/pedals is a joystick with many axes and buttons, and FFB in just one axis. This path should be easier.

lestcape, 2025-01-12:
Yes, @berarma, I agree, in a general sense it's not easy, I just thought that within the complicated stuff, this would be the place where it would be less complicated to do it. I also understand your point and I think you're right in trying to keep the driver only with what is really necessary, so that it actually works cleanly and can be brought to the kernel at some point.

Perhaps where you should make some adjustments is in your oversteer application, to decouple the pedals from the application, in case the user does not connect them to the wheel (maybe it requires the driver to report that there are no pedals in some way). This would send the clearest message, at least for me, that really the driver and the application only handle what comes from the connection of the wheel device and not anything else.