berarma/new-lg4ff #124: DKMS Install not working

OWL4C, 2025-04-11:
When installing with dkms i get the error that the same module cannot be installed more than once (once there is a full install, then another, which fails)

It loads fine, and even registers when i plug in the wheel (dmesg: Force feedback support for Logitech Gaming Wheels) but Oversteer registers no ff support and games do neither. When installing with sudo make install / sudo make load everything works perfectly, and the dmesg has an additional line about high res timers: Hires timer: period = 2 ms

berarma, 2025-04-11:
I pushed a commit recently trying to fix this issue. It just needs testing.

Hadrianneue, 2025-04-14:
not sure if related but i'm having a similar issue, works with kernel 6.14 but not with 6.15, i'm using the aur script to build it

```
DKMS (dkms-3.1.7) make.log for new-lg4ff/0.3.3.r40.g7b7c192 for kernel 6.15.0-rc1-1-mainline (x86_64)
seg 14 abr 2025 00:25:32 -03

Building module(s)
# command: make -j16 KERNELRELEASE=6.15.0-rc1-1-mainline KVERSION=6.15.0-rc1-1-mainline
make -C /lib/modules/6.15.0-rc1-1-mainline/build M=$PWD
make[1]: Entering directory '/usr/lib/modules/6.15.0-rc1-1-mainline/build'
make[2]: Entering directory '/var/lib/dkms/new-lg4ff/0.3.3.r40.g7b7c192/build'
  CC [M]  hid-lg.o
  CC [M]  hid-lg4ff.o
  CC [M]  hid-lgff.o
  CC [M]  hid-lg2ff.o
  CC [M]  hid-lg3ff.o
hid-lg4ff.c: In function ‘lg4ff_init’:
hid-lg4ff.c:2478:9: error: implicit declaration of function ‘hrtimer_init’; did you mean ‘hrtimers_init’? [-Wimplicit-function-declaration]
 2478 |         hrtimer_init(&entry->hrtimer, CLOCK_MONOTONIC, HRTIMER_MODE_REL);
      |         ^~~~~~~~~~~~
      |         hrtimers_init
make[4]: *** [/usr/lib/modules/6.15.0-rc1-1-mainline/build/scripts/Makefile.build:203: hid-lg4ff.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/lib/modules/6.15.0-rc1-1-mainline/build/Makefile:2006: .] Error 2
make[2]: *** [/usr/lib/modules/6.15.0-rc1-1-mainline/build/Makefile:248: __sub-make] Error 2
make[2]: Leaving directory '/var/lib/dkms/new-lg4ff/0.3.3.r40.g7b7c192/build'
make[1]: *** [Makefile:248: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.15.0-rc1-1-mainline/build'
make: *** [Makefile:5: default] Error 2

# exit code: 2
# elapsed time: 00:00:01
```

berarma, 2025-04-14:
> not sure if related but i'm having a similar issue, works with kernel 6.14 but not with 6.15, i'm using the aur script to build it
>
> ```
> DKMS (dkms-3.1.7) make.log for new-lg4ff/0.3.3.r40.g7b7c192 [...]

This is a different issue.