berarma/new-lg4ff #127: DKMS fails to install on Linux 6.15 kernel

kodatarule, 2025-05-29:
Hello,

After the update to kernel 6.15 it fails to install correctly in DKMS, I receive the following errors:

```
Error! The module/version combo: new-lg4ff/0.3.3.r42.g2092db1 is not located in the DKMS tree.
error: command failed to execute correctly
(1/1) reinstalling new-lg4ff-dkms-git                                                            [--------------------------------------------------------] 100%
Deprecated feature: CLEAN (/usr/src/new-lg4ff-0.3.3.r42.g2092db1/dkms.conf)
```

This is with the AUR package, clean re-built so it updates.

Operating System: CachyOS Linux 
KDE Plasma Version: 6.3.5
KDE Frameworks Version: 6.14.0
Qt Version: 6.9.0
Kernel Version: 6.15.0-2-cachyos (64-bit)
Graphics Platform: Wayland
Processors: 16 × AMD Ryzen 7 5800X3D 8-Core Processor
Memory: 31,2 GiB of RAM
Graphics Processor: NVIDIA GeForce RTX 3090

thecas, 2025-06-02:
Also got problems on NixOS with kernel version 6.15, maybe some of my logs help.

`
Running phase: unpackPhase
@nix { "action": "setPhase", "phase": "unpackPhase" }
unpacking source archive /nix/store/2k8mj280ig801rkfw06pwplgdaf81glw-source
source root is source
Running phase: patchPhase
Running phase: unpackPhase
@nix { "action": "setPhase", "phase": "unpackPhase" }
unpacking source archive /nix/store/2k8mj280ig801rkfw06pwplgdaf81glw-source
source root is source
Running phase: patchPhase
@nix { "action": "setPhase", "phase": "patchPhase" }
Running phase: updateAutotoolsGnuConfigScriptsPhase
@nix { "action": "setPhase", "phase": "updateAutotoolsGnuConfigScriptsPhase" }
Running phase: configurePhase
@nix { "action": "setPhase", "phase": "configurePhase" }
no configure script, doing nothing
Running phase: buildPhase
@nix { "action": "setPhase", "phase": "buildPhase" }
build flags: SHELL=/nix/store/xy4jjgw87sbgwylm5kn047d9gkbhsr9x-bash-5.2p37/bin/bash KVERSION=6.15.0 KDIR=/nix/store/gl4zb4minm4ddaz31lhsx2rr964z1gjl-linux-6.15-dev/lib/modules/6.15.0/build
make -C /nix/store/gl4zb4minm4ddaz31lhsx2rr964z1gjl-linux-6.15-dev/lib/modules/6.15.0/build M=$PWD
make[1]: Entering directory '/nix/store/gl4zb4minm4ddaz31lhsx2rr964z1gjl-linux-6.15-dev/lib/modules/6.15.0/build'
make[2]: Entering directory '/build/source'
  CC [M]  hid-lg.o
  CC [M]  hid-lg4ff.o
hid-lg4ff.c:462:6: warning: no previous prototype for 'lg4ff_send_cmd_with_id' [-Wmissing-prototypes]
  462 | void lg4ff_send_cmd_with_id(struct lg4ff_device_entry *entry, u8 *cmd, u8 id) {
      |      ^~~~~~~~~~~~~~~~~~~~~~
hid-lg4ff.c:480:6: warning: no previous prototype for 'lg4ff_send_cmd' [-Wmissing-prototypes]
  480 | void lg4ff_send_cmd(struct lg4ff_device_entry *entry, u8 *cmd)
      |      ^~~~~~~~~~~~~~
hid-lg4ff.c:498:6: warning: no previous prototype for 'lg4ff_update_slot' [-Wmissing-prototypes]
  498 | void lg4ff_update_slot(struct lg4ff_slot *slot, struct lg4ff_effect_parameters *parameters)
      |      ^~~~~~~~~~~~~~~~~
hid-lg4ff.c:1108:5: error: redefinition of 'lg4ff_adjust_input_event'
 1108 | int lg4ff_adjust_input_event(struct hid_device *hid, struct hid_field *field,
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from hid-lg4ff.c:22:
hid-lg4ff.h:15:19: note: previous definition of 'lg4ff_adjust_input_event' with type 'int(struct hid_device *, struct hid_field *, struct hid_usage *, s32,  struct lg_drv_data *)' {aka 'int(struct hid_device *, struct hid_field *, struc>
   15 | static inline int lg4ff_adjust_input_event(struct hid_device *hid, struct hid_field *field,
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~
hid-lg4ff.c:1134:5: error: redefinition of 'lg4ff_raw_event'
 1134 | int lg4ff_raw_event(struct hid_device *hdev, struct hid_report *report,
      |     ^~~~~~~~~~~~~~~
hid-lg4ff.h:17:19: note: previous definition of 'lg4ff_raw_event' with type 'int(struct hid_device *, struct hid_report *, u8 *, int,  struct lg_drv_data *)' {aka 'int(struct hid_device *, struct hid_report *, unsigned char *, int,  str>
   17 | static inline int lg4ff_raw_event(struct hid_device *hdev, struct hid_report *report,
      |                   ^~~~~~~~~~~~~~~
hid-lg4ff.c: In function 'lg4ff_alternate_modes_store':
hid-lg4ff.c:1651:67: error: 'lg4ff_no_autoswitch' undeclared (first use in this function)
 1651 |         if (target_product_id == USB_DEVICE_ID_LOGITECH_WHEEL && !lg4ff_no_autoswitch) {
      |                                                                   ^~~~~~~~~~~~~~~~~~~
hid-lg4ff.c:1651:67: note: each undeclared identifier is reported only once for each function it appears in
hid-lg4ff.c: In function 'lg4ff_handle_multimode_wheel':
hid-lg4ff.c:2176:14: error: 'lg4ff_no_autoswitch' undeclared (first use in this function)
 2176 |             !lg4ff_no_autoswitch) {
      |              ^~~~~~~~~~~~~~~~~~~
hid-lg4ff.c: At top level:
hid-lg4ff.c:2225:5: error: redefinition of 'lg4ff_init'
 2225 | int lg4ff_init(struct hid_device *hid)
      |     ^~~~~~~~~~
hid-lg4ff.h:19:19: note: previous definition of 'lg4ff_init' with type 'int(struct hid_device *)'
   19 | static inline int lg4ff_init(struct hid_device *hdev) { return -1; }
      |                   ^~~~~~~~~~
hid-lg4ff.c: In function 'lg4ff_init':
hid-lg4ff.c:2429:9: error: implicit declaration of function 'hrtimer_init'; did you mean 'hrtimers_init'? [-Wimplicit-function-declaration]
 2429 |         hrtimer_init(&entry->hrtimer, CLOCK_MONOTONIC, HRTIMER_MODE_REL);
      |         ^~~~~~~~~~~~
      |         hrtimers_init
hid-lg4ff.c: At top level:
hid-lg4ff.c:2444:5: error: redefinition of 'lg4ff_deinit'
 2444 | int lg4ff_deinit(struct hid_device *hid)
      |     ^~~~~~~~~~~~
hid-lg4ff.h:20:19: note: previous definition of 'lg4ff_deinit' with type 'int(struct hid_device *)'
   20 | static inline int lg4ff_deinit(struct hid_device *hdev) { return -1; }
      |                   ^~~~~~~~~~~~
make[4]: *** [/nix/store/gl4zb4minm4ddaz31lhsx2rr964z1gjl-linux-6.15-dev/lib/modules/6.15.0/source/scripts/Makefile.build:203: hid-lg4ff.o] Error 1
make[3]: *** [/nix/store/gl4zb4minm4ddaz31lhsx2rr964z1gjl-linux-6.15-dev/lib/modules/6.15.0/source/Makefile:2003: .] Error 2
make[2]: *** [/nix/store/gl4zb4minm4ddaz31lhsx2rr964z1gjl-linux-6.15-dev/lib/modules/6.15.0/source/Makefile:248: __sub-make] Error 2
make[2]: Leaving directory '/build/source'
make[1]: *** [/nix/store/gl4zb4minm4ddaz31lhsx2rr964z1gjl-linux-6.15-dev/lib/modules/6.15.0/source/Makefile:248: __sub-make] Error 2
make[1]: Leaving directory '/nix/store/gl4zb4minm4ddaz31lhsx2rr964z1gjl-linux-6.15-dev/lib/modules/6.15.0/build'
make: *** [Makefile:5: default] Error 2
`

berarma, 2025-06-02:
Could you report this to the AUR packager? It might need to be fixed there.

kodatarule, 2025-06-03:
> Could you report this to the AUR packager? It might need to be fixed there.

I will try to report it later, but apologies - wanted to correct myself that the module actually installs but complains about a previous version leftovers, other than that can confirm that it still works in 6.15.
Will close this here.

closed 2025-06-03 by kodatarule