berarma/new-lg4ff #130: [Feature Request] Gear shift debounce time / Pedal deadzone / Pedal sensitivity

Toquita3D, 2025-07-08:
Hi there!

I'm wondering if it would be possible to have these features. I have almost all racing simulators installed and only ACC allows for gear shift debouncing time adjustments. My G29 is having double downshift in Linux (didn't happen on Windows).

I'm probably asking too much, but would it be possible to have pedal deadzones (bottom/top) and pedal sensitivity as well? At least for the brake.

It would be awesome if Linux had these accessible via programs such as Oversteer.

Thank you for the amazing work you've done so far.

mksafavi, 2025-07-24:
My G29 recently developed the same issue on the upshift paddle. I tried an older version as well but given that only one button has this problem I was pretty sure that it's a hardware issue.

Bending the shifter microswitch spring outward almost fixed it for me, but I still had to set the debounce delay higher in game settings. 
I assume if a game doesn't have debouncing built in this will be a bigger problem.

I guess the only way of fully fixing it, is to replace the switches.

Toquita3D, 2025-07-24:
@mksafavi I opened my switch and followed [this tutorial](https://www.youtube.com/watch?v=ogjFYj_biUc) and it's now working perfectly.

berarma, 2025-07-24:
Great info, thank you both!